sypstraw/rpi4-osdev #7: UART minimal files and config on SD-Card, e.g. gpu_mem=???

no-go, 2021-03-05:
Nice tutorial, but please say a bit more about the bare minimum of files and `config.txt`

I need on my card this minimum for the UART example:

- important `bxm2711-rpi-4-b.dtb` (46.5 kB)
- important `config.txt` (125 byte)
- my `kernel8.img` (1.9 KB)
- the `start4.elf` (2.2 MB) = GPU firmware

The minimal content of `config.txt`:
````
[all]
core_freq_min=500
gpu_mem=128
````
A different gpu_mem like 64, 32 or 16 seams not to start the kernel (?)

isometimes, 2021-03-05:
Thanks for the feedback. I must confess that I never looked to minimise the set of files on the SD card (but I am intrigued now, so will look into it). I simply stuck with what the Raspberry Pi Imager put on there and just replaced `kernel8.img` with my own. I expect you'll need to at least add `fixup4.dat` to your list.

As for `config.txt`, mine looks like this and the UART runs just fine:

```c
hdmi_group=1
hdmi_mode=16
core_freq_min=500
```

You shouldn't need a `gpu_mem` line at all. Are you running on a standard Raspberry Pi 4 Model B?

no-go, 2021-03-05:
Hi, thanks for your answer. It is a "normal" Pi4 B with 4GB ram. Maybe there are early Hardware versions and I use a different/never EEPROM  (`recovery.img`), `start4.elf`, `fixup4.elf` and DTB file.

The miniuart example really works without `fixup4.elf` on my pi and I really need exact this `gpu_mem` line. It is ugly, that these elf files coming from Broadcom are closed source. I want to understand the whole boot process and that is quiet hard.

isometimes, 2021-03-08:
Yes - it's annoying that we can't strip right back... It stems from the fact that when the Raspberry Pi powers on, the ARM CPU is off and the GPU is on.

The GPU runs the first bootloader, which is held in ROM on the SoC (System on Chip). This is similar to the BIOS in conventional PCs.

This bootloader reads the SD card and loads the second bootloader from `bootcode.bin`.

The second bootloader then reads the GPU firmware, also from the SD card, named `start.elf`.

Finally, `start.elf` reads `kernel.img` and allows the ARM CPU to execute it.

no-go, 2021-03-08:
The hardware is open enough to explain some base OS techniques to students. That's why I try to migrate a minimalistic BIOS x86 32bit based OS (c++ and some asm) to aarch64 (primary the Pi4). I think the PL0/1 and paging instead of Ring1/0 and "sections and GDT" will be the next big trouble in my tasklist ;-D

isometimes, 2021-03-08:
Have fun and best wishes for it! :) I actually started with an x86 minimal OS before I bought the Pi. So much more enjoyable on AArch64 though, I must say...

closed 2021-03-08 by isometimes