sypstraw/rpi4-osdev #8: MMU and memory support?

wooshifu, 2021-03-16:
Are there any plans about `MMU` and physic & virtual memory management?

isometimes, 2021-03-16:
Hi,

I'm currently working on "part12" which will add a lot more graphics capabilities (all still CPU/GPU). That's required me to start implementing some ultra-simple heap management (still just physical memory) - basically malloc & free in `lib/mem.c`. I'm avoiding MMU/virtual memory for now as it gets very complicated quite quickly.

After I've got the graphics library finished, I plan to consolidate all my work so far.

Have you bztsrc's tutorial on virtual memory? https://github.com/bztsrc/raspi3-tutorial/tree/master/10_virtualmemory. Some changes may be required to make this work on RPi4... It's something I'm looking forward to, but I want to finish my graphics work first ;-)

Hope it helps!

A

wooshifu, 2021-03-18:
Thanks for your reply 👍

closed 2021-03-18 by wooshifu