sypstraw/rpi4-osdev #13: Low Peripheral mode?

timonsku, 2021-05-16:
I was a bit confused about the `PERIPHERAL_BASE` being set to `0xFE000000` in these examples as the BCM2711 doc is listing a different base address in the docs. Then I stumbled over this sentence in the beginning of the docs:

> So a peripheral described in this document as being at legacy address 0x7Enn_nnnn is available in the
> 35-bit address space at 0x4_7Enn_nnnn, and visible to the ARM at 0x0_FEnn_nnnn if Low Peripheral
> mode is enabled.

So I'm assuming you are using this Low Peripheral mode? The docs say you need to specifically enable it though but do not seem to talk further about it. Where are you enabling this mode or is the wording confusing and its always enabled by default?
Also what is the use of it?

isometimes, 2021-05-16:
Yeah - low peripheral mode is the default boot environment. If you want to set high peripheral mode, you can do it explicitly in config.txt using arm_peri_high=1 (source: https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md).

You'll need to do quite a bit of refactoring to make the kernel in this tutorial support that though...

All the best,
Adam

isometimes, 2021-05-16:
Essentially "Low Peripheral mode" maps the peripherals over the last 64mb of RAM.

People might wish to enable High Peripheral mode (full 35-bit address map) so as to avoid "losing" that last 64mb of RAM. There are various side effects, however, of doing this and it would require some refactoring of the kernel (even in this simple tutorial) to make it work.

This thread does a good job of explaining the nuances: https://github.com/raspberrypi/firmware/issues/1374.

Hope it helps!
A

timonsku, 2021-05-16:
Oh I see! Thanks a lot of the info, that cleared up a lot for me.

closed 2021-05-16 by timonsku

guichristmann, 2022-02-20:
I was confused by the exact same issue when reading the BCM2711 Peripherals manual. Maybe it'd be nice to add little note regarding this in Part 4?

Amazing resource by the way. Appreciate it a lot, and already looking forward to the next steps in this journey.

isometimes, 2022-02-20:
Thanks @guichristmann, it's a good point! I've added a note at the beginning of part4-miniuart as you suggested 👍