sypstraw/rpi4-osdev #14: Undefined reference to 'main'

armcgill, 2021-06-07:
Hello! 
I have been jumping into the world of OSDev and stumbled across your amazing write ups. 

I am on step two creating the make file and I keep getting the error: 

(.text.boot+0x34): undefined reference to 'main'
make: *** [Makefile:15: kernel18.img] Error 1 

I am trying to figure out if the issue is in my linker.ld file or the Makefile itself. It seems to properly make the boot.o and kernel.o files but fails when trying to link them into one file. 

I have checked multiple times my linker.ld file matches yours as well as the Makefile (minus the file path to the cross compiler). 

Any suggestions of what I can try? Thanks for your help and for taking the time to make these tutorials!  

[image]
[image]

isometimes, 2021-06-07:
Hi,

Great to hear from you! This error is what I'd expect to see if you didn't have a `main()` function defined in _kernel.c_.

Can you post the contents of your _kernel.c_ file? If you're strictly going through the tutorial then this should read as follows for part2:

```c
void main() {
   while (1);
}
```

It's very unexciting, it literally just spinning in an infinite loop doing nothing! I promise it gets more interesting as you go along though!

All the best,
Adam

armcgill, 2021-06-07:
Thank you so much for responding! After attempting to compile it again (with no changes to the code), it worked with no issue and booted on my Pi! I am not sure what was giving me the error for so long.  I am looking forward to continuing through the rest of your write up.

closed 2021-06-07 by armcgill

isometimes, 2021-06-07:
That's great news! Enjoy the journey ☺️