sypstraw/rpi4-osdev #15: trick to use wav file (part-9 sound)

no-go, 2021-08-11:
Hi,

I am actually testing DMA with a wav file (unsigned 8bit, stereo, 44.1kHz). I saw this in config.txt (risc os):

> ramfsfile=CMOS
> ramfsaddr=0x508000

After some tests, I did this (address above 8F0 0000 will not work):

> ramfsfile=audio.wav
> ramfsaddr=0x1C0000

Because wav files (and header) are not so magic, I read the data like this (for DMA):

> char * data = (char *) 0x1C0000;
> unsigned size = *((unsigned *) (0x1C0000+40));
> data += 44;
> for (unsigned i=0; i<size; i++) *(safe+i) = *(data+i);

If someone wants to split sound and kernel, this would be a nice option ;-)

Links:

- https://www.riscosopen.org/wiki/documentation/show/Software%20information:%20Raspberry%20Pi:%20Firmware
- http://soundfile.sapp.org/doc/WaveFormat/

isometimes, 2021-08-11:
Oh, that's neat!

The (painful) alternative is to write the SDIO stuff required to get files off the SD card. That requires a whole world of code to read the FAT filesystem etc. (which is probably copyright M$!)

This is a rather great middle ground. Thanks for sharing!

I will try to add it to the docs at some point and credit you :) Unfortunately I've been distracted from this project for a little while - but my plan is to come back to it very soon. It's so much fun!

paulwratt, 2022-07-04:
> That requires a whole world of code to read the FAT filesystem etc. (which is probably copyright M$!)

No, there are a couble of RPi baremetal FAT on USB pieces of code in my repository list (which means it is/was here on GitHub somewhere), and I think at least one is assembler

isometimes, 2022-07-05:
@paulwratt I must confess that I've got code for reading the FAT filesystem off the SD card... I just don't like/trust it enough to publish it! It was a pretty limited & clunky implementation as a proof of concept... I agree that this is not needed if we use the `ramfs` though. Weirdly I couldn't make @no-go's suggestion work. I suspect my brain was just on holiday that day (it happens!) and I haven't yet revisited it. I still really like the idea though which is why I left the issue open as a burning reminder 😄

closed 2023-03-30 by isometimes