sypstraw/rpi4-osdev #23: A way to implement C++

3c4ng1s, 2021-11-28:
Hi, I’m not so sure if I’m doing this right and if my question is an issue at all, but just in case I’m leaving it here. I have read your code and I have enough knowledge to understand most part of it and actually get what is doing, but I would love to implement a kernel in cpp. Somehow I’m assuming at first that it will be connected to the compiler and compilation process, but I wouldn’t know how it should actually go.

isometimes, 2021-11-28:
Hi,

If you're using clang as your compiler, you simply use clang++ instead. Or if you're using gcc, use g++ instead.

Main changes (beyond the code) will be to the Makefile therefore, but the linking process should be the same.

I'd start simply and try to get part2 in pure C++ if you want. I'm not a C++ programmer, but I imagine the switch will be very easy. Circle is a good C++ OS/kernel that runs on the RPi - you might be able to take your cues from some of that too...

Thanks,
Adam

isometimes, 2021-11-28:
Circle is here BTW: https://github.com/rsta2/circle

A

closed 2021-11-29 by isometimes

3c4ng1s, 2021-11-30:
yo leí sobre circulo pero me gustaria hacer algo aún más básico por objetivos de aprendizaje

isometimes, 2021-11-30:
I wish you all the best for it :) You've made a great decision to go on this journey! I agree that C++ would ultimately end up in a tidier codebase... Maybe one day I'll give it a go!!