sypstraw/rpi4-osdev #27: Error with Makefile on Windows

BowersIndustry, 2022-05-10:
Hello! I've been following your tutorial, but when I copied the Makefile.gcc.windows, I renamed it just Makefile so I can type "make". But when I did that,, if gave me this error. I don't know if I'm using the wrong version of GCC or not. I made sure I had the ARM version.
`del kernel8.elf *.o *.img
/bin/sh: line 1: del: command not found
make: *** [Makefile:22: clean] Error 127
PS E:\Repos\KemyonOS\KemyonOS\KemyonOS>`

isometimes, 2022-05-10:
Hi!

It looks like you're not using Windows since you're running /bin/sh...

"del" is a command in Windows command prompt, but not in WSL which is a Linux subsystem. make can't find it because the Linux equivalent is "rm".

If you're building on Linux you'll need to use the normal gcc Makefile. Does it make sense?

Thanks,
Adam

BowersIndustry, 2022-05-10:
I'm using Windows 10. I'm also using VSC to write the code. Does that have anything to do with it?

isometimes, 2022-05-11:
So it looks like VSC is shelling out to /bin/sh which won't find a "del" command that "clean" needs. You'll need to use "/bin/rm" instead.

Maybe try with the normal gcc Makefile first (without the .windows extension) before you get into editing the Makefile?

Thanks,
Adam

isometimes, 2022-05-12:
Sorry, I don't use VSC (never have) and so I'm not sure how to help. I've tested the Makefiles (both .gcc using WSL, and .gcc.windows) on my own Windows box using just the Arm compiler from the command line (no environment variables needed) and both build just fine. I have no idea why the "del" command seemingly can't be found on your box. This feels environmental, and so I'm sorry I can't really help :(

closed 2022-05-12 by isometimes