sypstraw/rpi4-osdev #29: Makefile path to LLVM differs

greenonline, 2022-07-03:
Minor point, which *might* be of use to others (possibly)...

I found that I had to change a line in the Makefile (on Catalina - may apply to others), setting the path for `LLVMPATH`:

```none
#LLVMPATH = /opt/homebrew/opt/llvm/bin 
LLVMPATH = /usr/local/opt/llvm/bin
```

I *may* also have had to run (but I'm not convinced that it was absolutely necessary):

```none
export LDFLAGS="-L/usr/local/opt/llvm/lib -Wl,-rpath, /usr/local/opt/llvm/lib" 
export CPPFLAGS="-I/usr/local/opt/llvm/include" 
export LDFLAGS="-L/usr/local/opt/llvm/lib" 
echo 'export PATH="/usr/local/opt/llvm/bin:$PATH"' >> ~/.zshrc 
. ~/.zshrc
```

This last batch of commands was advised by the output of 

```none
brew install llvm
```

so I ran them just in case.

paulwratt, 2022-07-04:
That `LLVMPATH` suggest that Catalina has changed the location for HomeBrew installations.

I would be interested to see what `LLVMPATH` is set to on other BSD platforms, as it may be accurate to update the tutorial with that exact example

isometimes, 2022-07-05:
Agreed, it _is_ useful to call out things which have helped you progress in your particular environment. I suspect some of my setup is non-standard, particularly as I got an M1 in the early days when some things I needed were unsupported, provoking workarounds. I'm going to close this issue without updating the docs, but thanks for the useful insights.

closed 2022-07-05 by isometimes