sypstraw/rpi4-osdev #37: Raspberry Pi 4 kernel load address

tdh8316, 2022-07-10:
Hi! first of all, thank you for writing this guide.
I followed everything in part 2-building and wrote my own kernel.c due to I have no UART connector.
```c
// kernel.c
#define BCM2711_PERI_BASE_ADDR 0xFE000000
#define GPIO_BASE_ADDR BCM2711_PERI_BASE_ADDR + 0x200000

#define OUT_PIN 21

volatile unsigned *m_pGPIO;

void setGPIOModeInput(int pin)
{
    const int reg = pin / 10;
    const int shift = (pin % 10) * 3;
    *(m_pGPIO + reg) = ((*(m_pGPIO + reg) & ~(7 << shift)) | (0 << shift));
}

void setGPIOModeOutput(int pin)
{
    setGPIOModeInput(pin);
    const int reg = pin / 10;
    const int shift = (pin % 10) * 3;
    *(m_pGPIO + reg) = ((*(m_pGPIO + reg) & ~(7 << shift)) | (1 << shift));
}

inline __attribute__((always_inline)) void setGPIO(int pin)
{
    *(m_pGPIO + 7) = 1 << pin;
}
inline __attribute__((always_inline)) void clrGPIO(int pin)
{
    *(m_pGPIO + 10) = 1 << pin;
}

int main(void)
{
    m_pGPIO = (volatile unsigned *)GPIO_BASE_ADDR;

    setGPIOModeOutput(OUT_PIN);

    while (1)
    {
        setGPIO(OUT_PIN);
        clrGPIO(OUT_PIN);
    }
}

```
And I made the latest Raspbian-Lite (64bit) micro SD card using raspberry pi imager. Then replaced the `kernel8.img`.

I expected GPIO21 (40 for hardware) to generate spikes, but it didn't.
I'm not even sure whether the boot sequence is done; I found [this post](https://forums.raspberrypi.com/viewtopic.php?t=328000#p1966797) on the forum. They say the kernel load address is changed to 0x200000.

Is this guide affected by that post? If not, I'll assume that the `kernel.c` contains problems.

isometimes, 2022-07-11:
Hi,

Thanks for your message. The kernel load address should still be `0x80000`.

I suspect you have an issue with your _kernel.c_.

All the best,
Adam

isometimes, 2022-07-11:
You might want to take a look at how you set the GPIO to be an output. I don't think you need to respect the current value by using OR - it may lead to unexpected results.

I might be wrong, but it's worth a look!

tdh8316, 2022-07-12:
Wow, I didn't expect you to provide an example for me. I just realized that my code was wrong, that the kernel load address is still there. Thanks for your favor! You're the kindest person I've ever experienced. 

Have a great day,
Best regards!