sypstraw/rpi4-osdev #38: Building to test on QEMU/KVM

Tall-Lobster, 2022-10-22:
I wanted to build this project to test it on QEMU instead of having to switch back and forth between raspberry pi sd cards. I was wondering if you would give me a hand with it?

isometimes, 2022-10-22:
Are you targeting a RPi4 QEMU image? If so, at 32-bit or 64-bit?

I don't think QEMU has OOB support for RPi4 yet, to be honest. There may be somebody out there with a project to patch it, but that's not an insignificant piece of work AFAIK (never used QEMU).

You may have more luck building a kernel that receives an image streamed over serial and boots from that. Many have done this to good effect. That solves the SD card issue! It's still an unpleasant experience in my opinion though.

Let me know why you think QEMU is the right answer and perhaps I can have a dig around?

paulwratt, 2022-10-24:
if you are willing to use docker containers, then this will help you:

https://github.com/M0Rf30/simonpi

If you check that users repos, you might find some other interesting RPi4 Qemu things too

isometimes, 2022-10-26:
I wish you all the best of luck with QEMU. It's not my area of expertise - I prefer real hardware! I think @paulwratt's comment should give you a good start though. Do share if you get sometime working ☺️

closed 2022-10-26 by isometimes