sypstraw/rpi4-osdev #44: Question: USB Support

suityou01, 2023-08-17:
In your excellent tutorials you mention you pursued BLE as USB was much more difficult. Could you sketch out the steps, as you see it, even if from a high level, on how to go about adding USB support to this Kernel?

I note you added the HCD file from the bluetooth firmware to your kernel. Is there a similar firmware we can leverage for the USB chip on these boards? Presumably this is a client server architecture (or host, client)?

isometimes, 2023-08-17:
It's a great challenge and one that would create a whole world of possibilities on bare metal. I will have a look into the link you sent and whether it could accelerate this kind of work on the RPi4 for us. I'd be excited about getting USB up to enable access to the onboard Ethernet controller rather than relying on an SPI peripheral to do that (easier interface). Then there's obvious stuff like being able to connect a keyboard/pointing device etc.

It's been a while since I have updated the tutorial and so maybe this could be a good (but challenging) next step? 🤔 

Let me write when I have some idea of feasibility!

Thanks for the inspiration/nudge...

suityou01, 2023-08-18:
Looks like tinyusb does not have support for xhci. It could be added though and there are people talking about how. This git was linked >> https://github.com/cherry-embedded/CherryUSB/tree/master/port/xhci

isometimes, 2023-08-18:
Hmmmmnn, well USB 2.0 would be a great start - USB 3.0 is just the icing on the cake! 🍰

suityou01, 2023-08-18:
Agreed. I found this on OSDev which purports to be the concise explainer of USB 2.0 from the spec
https://wiki.osdev.org/Universal_Serial_Bus

I shall have a read and revert with findings.

suityou01, 2023-08-19:
So it appears we will need some memory management, chiefly spinlocks and dma. This can be implemented as separate chapters I would have thought and would then segue into the USB piece.

I also found this >> https://github.com/Chadderz121/csud/tree/master

isometimes, 2023-08-19:
Yup - and actually I implement DMA in the sound tutorial and touch on the spinlock concept in the multi core tutorial, so we're not a million miles off...

isometimes, 2023-08-23:
Also wonder if we couldn't mod this to support Raspberry Pi 4? https://github.com/rsta2/uspi

isometimes, 2023-08-25:
Nice! I have to say that I'm really targeting the Pi4 which has significant hardware differences... Seems non-trivial

suityou01, 2023-08-25:
Agree and my aim is aligned with yours. Qemu supports the PI3 out of box (there are Pi4 versions out there that I will try in time). I just want to get to the point where I have a barebones OS running on QEMU with something Pi shaped enumerating the PCIe bus from my native code. The same code should in theory port to the Pi4 and once we can get the address of the USB root controller from the PCIe bus then we are off and running.

suityou01, 2023-08-25:
Fell at the second hurdle with Qemu. Link to issue >>
https://gitlab.com/qemu-project/qemu/-/issues/1845

isometimes, 2023-10-09:
I think we close this for now. I am excited to see what will be possible with RPi5... ;)

closed 2023-10-09 by isometimes