sypstraw/rpi4-osdev #45: Timer 3 bar not fully painted in Part 13 (Interrupts)

petemoore, 2023-11-27:
Thanks for the write up!

I tried out part 13, which worked as expected, other than the "Timer 3" bar never filled on my Raspberry Pi 400.

[image]

Perhaps this is ok, but since it differs from [your screenshot](https://raw.githubusercontent.com/isometimes/rpi4-osdev/master/part13-interrupts/images/13-interrupts-running.jpg) I thought I'd let you know. Feel free to close if this isn't an issue.

This is using latest raspberry pi firmware (commit 93d3f791a28ab369a62d2baf06035fc533b00864) and my config.txt looks like:

```
core_freq_min=500
disable_commandline_tags=1
kernel_old=1
arm_64bit=1
hdmi_group=1
hdmi_mode=16
```

isometimes, 2023-11-30:
Hi Pete,

I have seen that behaviour from time to time too. So it's hard to dismiss it and close it as an issue. Requires some serious debug I think. Unfortunately I've packed away my RPi4 now as I start to think about how to transition this to RPi5. Admittedly that's a lofty goal considering the hardware has changed so significantly! If I get too frustrated, I will come back and take another look at this. Have you any ideas what you think might be causing it?

Thanks,
Adam

petemoore, 2023-12-05:
Something occurs to me here:

```
void handle_irq() {
    unsigned int irq = REGS_IRQ->irq0_pending_0;

    while(irq) {
        if (irq & SYS_TIMER_IRQ_1) {
            irq &= ~SYS_TIMER_IRQ_1;

            handle_timer_1();
        }

        if (irq & SYS_TIMER_IRQ_3) {
            irq &= ~SYS_TIMER_IRQ_3;

            handle_timer_3();
        }
    }
}
```

If you get an interrupt that is not from SYS_TIMER_IRQ_1 or SYS_TIMER_IRQ_3, this loop will loop forever.

Now normally, this shouldn't be a problem, because only these two interrupts are enabled here:

```
void enable_interrupt_controller() {
    REGS_IRQ->irq0_enable_0 = SYS_TIMER_IRQ_1 | SYS_TIMER_IRQ_3;
}
```

However, in kernel.c we have

```
enable_interrupt_controller();
irq_enable();
```

Perhaps enable_interrupt_controller hasn't fully taken effect when irq_enable() is called, and when interrupts are enabled, some other interrupt leaks into handle_irq that isn't one of the two you process there. It would then get stuck in an infinite loop.

This is just a hunch, I don't have my rpi400 with me. But I can imagine two mitigations here: one would be to modify handle_irq while loop from:

```
while (irq) {
```

to:

```
 while(irq & (SYS_TIMER_IRQ_1 | SYS_TIMER_IRQ_3)) {
```

Another mitigation could be to drop a barrier instruction between these two lines:

```
enable_interrupt_controller();

// put a barrier here - DSB?

irq_enable();
```

I haven't read up on barriers properly yet, not sure if DSB is the correct barrier for this situation, but it could be worth looking into.

I haven't actually tested this hypothesis, it was just an idea that came to me while browsing through code while at a cafe without my rpi400 with me!

Let me know if it helps though!

Pete

isometimes, 2023-12-05:
Great thinking - let me know if it works if you get back to your Pi before I get to mine! Thanks so much for your brain cycles on this 😀 Also, do feel free to submit a PR if you find it solves the problem - would be great to acknowledge you as a contributor!

babbleberry, 2024-02-06:
Looking into this today... weirdly I can't reproduce when building from WSL (Ubuntu) using gcc. However, when I build on MacOS using Clang, it's completely reproducible. Going to try your suggested fixes in both build environment now... Watch this space!

babbleberry, 2024-02-06:
OK, so inserting the `dsb sy` barrier works a charm on Clang...

babbleberry, 2024-02-06:
As does the more explicit while loop. I think that's my preferred solution to be honest... feels less "sledgehammer to crack a nut"! Will test on the Windows build environment and then commit... Thanks!!

babbleberry, 2024-02-06:
Decided to go with both suggested fixes in the end for belt & braces! All tested and committed - thanks again for your input. I'd be grateful if you could test in your environment when you have a moment :)

Thanks,
Adam

petemoore, 2024-02-06:
Hi Adam,
Apologies that I never got round to looking at this again - I'm really happy the fixes worked though! When I get a chance I'm happy to retest here too. Thanks for addressing the issue, and for creating and sharing this really nice project. :-)
Pete

babbleberry, 2024-02-08:
Frustratingly I have managed to repeat the bug even with both fixes in place concurrently! Back to the drawing board sadly... 😢

babbleberry, 2024-04-22:
I think a few of my more recent changes have resolved this. Going to close (again) for now!

closed 2024-04-22 by babbleberry