sypstraw/rpi4-osdev #46: Various Issues when creating the Makefile

EinSatzMitX, 2024-01-27:
I've had an issue where, when i was creating the makefile, i would get an Error Message similar to this:

einsatzmitx@DESKTOP-K54UIMD:/mnt/c/Users/Peter/Desktop/Additum2024/Rpi4OS$ make
/bin/rm kernel8.elf *.o *.img > /dev/null 2> /dev/null || true
C:/gcc-arm-10.3-2021.07-x86_64-aarch64-none-elf/gcc-arm-10.3-2021.07-x86_64-aarch64-none-elf/bin/aarch64-none-elf-gcc -Wall -O2 -ffreestanding -nostdinc -nostdlib -nostartfiles -c boot.S -o boot.o
make: C:/gcc-arm-10.3-2021.07-x86_64-aarch64-none-elf/gcc-arm-10.3-2021.07-x86_64-aarch64-none-elf/bin/aarch64-none-elf-gcc: No such file or directory
make: *** [Makefile:9: boot.o] Error 127

I've tried various ways of changing the GCCPATH, but none seemed to work, so I am here to ask for help
Thanks for the help in advance!

babbleberry, 2024-01-30:
Hi,

No worries - what operating system are you building on and what is the path of your GCC installation?

Thanks,
Adam

EinSatzMitX, 2024-01-31:
Hey,

Thanks for your comment but it turns out I'm just incredibly stupid. I don't really know  what I was doing when trying to install aarch64 via the link in the first readme but There's a much easier way using 'sudo apt install'.
Still thanks for your comment and in case anyone should suffer from the same stupidity as me, I'll leave this here.
Happy coding!

closed 2024-01-31 by EinSatzMitX