sypstraw/rpi4-osdev #48: Question: How did install aarch64?

EinSatzMitX, 2024-02-01:
Hey!

I've been following this tutoial for a few days now and I must say that it's become increasingly hardwr to follow along as you apparently installed aarch64 on a different way than I did. I installed it using  ''' sudo apt get '''
on WSL but through that i have to write my makefile content different and that's become very hard, as I am getting strange error messages that shouldn't come up at all.
I'd like to hear what you did different because it's probably really simple and I might have just forgotten to do it.
Thanks for your response and patience with me :D

EinSatzMitX, 2024-02-01:
ok so i just copied your stuff again .-. and i get this as an output:

einsatzmitx@DESKTOP-K54UIMD:/mnt/c/Users/Peter/Desktop/Additum2024/Rpi4OS$ make
/bin/rm kernel8.elf *.o *.img > /dev/null 2> /dev/null || true
make: *** No rule to make target 'boot.S', needed by 'boot.o'.  Stop.

babbleberry, 2024-02-02:
Intriguing! Can you share a copy of your Makefile please?

A

babbleberry, 2024-02-02:
As for how I got my environment up and running... from a clean WSL Ubuntu environment, follow this paragraph:

Download and unpack [Arm’s gcc compiler](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads). For reasons that I won’t go into here, you’ll need to use the “AArch64 ELF bare-metal target”. Since I’m using WSL on Windows 10 to emulate Ubuntu, I downloaded the x86_64 Linux hosted cross-compiler.

Also install GNU make - you’ll need it soon enough. Because I’m using WSL, for me that was simply a matter of typing sudo apt install make and entering my password.

EinSatzMitX, 2024-02-04:
Ok so it turns out that,  since I installed the Compiler on WSL and in my makefile i wrote 
'''
GCCPATH ?= aarch64-linux-gnu
'''
I would have to change my code from
'''
        $(GCCPATH)/aarch64-none-elf-gcc $(GCCFLAGS) -c boot.S -o boot.o
'''
To 
'''
        $(GCCPATH)-gcc $(GCCFLAGS) -c boot.S -o boot.o
'''
This resolved the wide variety of makefile errors i got and since that everything has been working alrighr

EinSatzMitX, 2024-02-04:
Ok so it turns out that,  since I installed the Compiler on WSL and in my makefile i wrote 
'''
GCCPATH ?= aarch64-linux-gnu
'''
I would have to change my code from
'''
        $(GCCPATH)/aarch64-none-elf-gcc $(GCCFLAGS) -c boot.S -o boot.o
'''
To 
'''
        $(GCCPATH)-gcc $(GCCFLAGS) -c boot.S -o boot.o
'''
This resolved the wide variety of makefile errors i got and since that everything has been working alright

babbleberry, 2024-02-04:
Glad to hear you've resolved the issue! Sorry I wasn't more helpful 😄

closed 2024-02-04 by babbleberry