sypstraw/rpi4-osdev #53: part9-sound clock registers

qzpt, 2024-05-22:
Hi,

Thanks for the interesting tutorial.

I am trying to run bare-metal I2S on RPi 4B. I am having some issues regarding register addresses, so I came along your tutorial.

In the tutorial, the clock base is https://github.com/babbleberry/rpi4-osdev/blob/02d190e8f7a8ef6ce2d72d42781500f97a3c4149/part9-sound/kernel.c#L6 as defined in the BCM2711 datasheet. However, I am struggling to find `BCM2711_PWMCLK_CNTL` and `BCM2711_PWMCLK_DIV` in the datasheet and there offset being 40 and 41, respectively. As in the datasheets the clock registers offset is starting from `0x70` to `0x84`. 
[image]

Any help would be appreciated.

Thanks

babbleberry, 2024-05-29:
Hi,

This is a very good point. However, you're looking at the General Purpose Clock rather than the PWM clock.

That being said, the PWM clock is actually at offset 0xA0 (CTL) and 0xA4 (DIV). It's undocumented in the RPi docs but can be found in the Broadcom docs and cited here: https://www.scribd.com/doc/127599939/BCM2835-Audio-clocks

Now I'm wondering why 40 and 41 work rather than 0xA0 and 0xA4! I think I need to experiment with this some more :)

babbleberry, 2024-06-04:
OK, I'm right on all counts (rare!)... The in-memory offsets are 0xA0 and 0xA4 respectively as per the docs (see my previous comment).

40 (0x28) is actually 0xA0 divided by 4, and 41 (0x29) is 0xA4 divided by 4. Why divide by 4? Well, 4 is the sizeof a volatile unsigned. So every time I add 1 to a volatile unsigned pointer, I'm actually moving it by 4 bytes.

Mystery solved!!!

closed 2024-06-04 by babbleberry