sypstraw/rpi4-osdev #54: Part 9 - Sound 16 bit audio

Preciseful, 2024-09-10:
Hey,
I've tried messing around with the code (the CPU sound more specifically), in order to get my audio running at 16 bit. I tried changing all values that I found to be relevant, and tried some other stuff. Is it possible even?
And if it is, how would I go on about doing it?

babbleberry, 2024-09-15:
Hi,

I must confess that I haven't looked at this. Perhaps you could share your code with comments and I'll be happy to point out anything I think might be causing the problem?

Thanks,
Adam

babbleberry, 2024-09-16:
Oh? I thought you said you'd "messed around" with the code? Just changing the bit resolution in the file won't be enough...

Preciseful, 2024-09-16:
I did mess around but it wasn't anything more than like.. some slight different values that I eventually reverted because it did not work. I also had no clue what I was doing